ethz-asl/ethzasl_icp_mapping #28: Support Clang

dreuter, 2015-03-04:
When using clang I get the following error:

CMake Error at libpointmatcher_ros/ethzasl_point_cloud_vtk_tools/CMakeLists.txt:42 (message):
  You need partial C++0x support for N2351, N2672, N1984.  Currently this
  build toolchain supports only GCC >= 4.4.  If your compiler supports these
  specifications, please send us a patch.

Please add clang to the list of supported compilers

pomerlef, 2015-03-11:
Hi there,

Unfortunately, I'm not install to test that. The solution would be to use those if statements in the `CMakeLists.txt`:

```
if ("${CMAKE_CXX_COMPILER_ID}" STREQUAL "Clang")
  # using Clang
elseif ("${CMAKE_CXX_COMPILER_ID}" STREQUAL "GNU")
  # using GCC
elseif ("${CMAKE_CXX_COMPILER_ID}" STREQUAL "Intel")
  # using Intel C++
elseif ("${CMAKE_CXX_COMPILER_ID}" STREQUAL "MSVC")
  # using Visual Studio C++
endif()
```

If you could make those modifications in your personal branch and send a pull request once it's working on your side, I could test on my system to make sure that nothing breaks and add your patch on the repository.

pomerlef, 2015-10-08:
Any follow up on that?

dreuter, 2015-10-15:
Sadly I did not have have time to test this for now... I just switched our project back from clang to gcc for now.

Actually I am waiting for CMake 3.1 to be available in Ubuntu 14.04 (but I guess this won't happen), because it has proper C++11 support using [target_compile_features](https://cmake.org/cmake/help/v3.1/command/target_compile_features.html).

I am aware, that we could just expand the list of checks for every single compiler, but it feels just very hacky. But maybe I will do it anyway, but sadly I can't promise when.