ethz-asl/ethzasl_icp_mapping #31: Issue with LaserScan frames in 2D mapper

fcolas, 2015-07-09:
There seem to be a frame consistency issue in `ethzasl_icp_mapper` for 2D scans.
When the mapper receives a  `LaserScan` message, it is first [transformed in a `DataPoints` object into `odomFrame`](https://github.com/ethz-asl/ethzasl_icp_mapping/blob/master/ethzasl_icp_mapper/src/mapper.cpp#L298), in order to account for the motion of the robot during the scan, then the `processCloud` method is called with this `DP` but [specifying the frame as `scanMsgIn.header.frame_id`](https://github.com/ethz-asl/ethzasl_icp_mapping/blob/master/ethzasl_icp_mapper/src/mapper.cpp#L300).

It seems inconsistent and, indeed, when playing the [example bag](https://github.com/ethz-asl/ethzasl_icp_mapping/blob/master/ethzasl_icp_mapper/launch/2D_scans/2D_mapper.launch#L5) we observe that the robot location doesn't move and is completely incoherent with the map (which seems fine). Changing `scanMsgIn.header.frame_id` into `odomFrame` when calling `processCloud` fixes this issue.

I can submit a pull request but it seems that this inconsistency was there [from the start](https://github.com/ethz-asl/ethzasl_icp_mapping/commit/16a21e68e0bcf7ffffd8c43ed61c0c64062ac1e1) so I'm not sure how it could have worked nor whether there could be specific cases that work only this way.

kruesip, 2015-07-09:
I think that this is the same problem that I tried to solve in #30 

IMO the point is that rosMsgToPointMatcherCloud() should not transform the points to the odomFrame, but to the scanner's frame at one specific point in time (namely the timestamp of the incoming message). In the 3D case we always have the incoming point clouds represented in the scanner frame, right? We never transform them to the odomFrame. To solve that issue I changed the "target_frame" and "target_time" in the transformPoint() method (see #30). Worked well in my experiments.

pomerlef, 2015-07-09:
Oh! It seems that I forgot about #30...

@fcolas Could you try #30? If it work on your side, I'll accept it.

fcolas, 2015-07-09:
Yep, #30 is a good fix.

closed 2015-07-09 by pomerlef