ethz-asl/ethzasl_icp_mapping #34: compilation error for indigo_devel branch

nitin5, 2015-08-06:
Hi 
I am facing compilation errors.
Following is the output of the catkin_make

nitin@nitinHP:~/catkin_ethzasl_icp$ catkin_make
Base path: /home/nitin/catkin_ethzasl_icp
Source space: /home/nitin/catkin_ethzasl_icp/src
Build space: /home/nitin/catkin_ethzasl_icp/build
Devel space: /home/nitin/catkin_ethzasl_icp/devel
Install space: /home/nitin/catkin_ethzasl_icp/install
####
#### Running command: "cmake /home/nitin/catkin_ethzasl_icp/src -DCATKIN_DEVEL_PREFIX=/home/nitin/catkin_ethzasl_icp/devel -DCMAKE_INSTALL_PREFIX=/home/nitin/catkin_ethzasl_icp/install -G Unix Makefiles" in "/home/nitin/catkin_ethzasl_icp/build"
####
-- The C compiler identification is GNU 4.8.2
-- The CXX compiler identification is GNU 4.8.2
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Using CATKIN_DEVEL_PREFIX: /home/nitin/catkin_ethzasl_icp/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/indigo
-- This workspace overlays: /opt/ros/indigo
-- Found PythonInterp: /usr/bin/python (found version "2.7.6") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/nitin/catkin_ethzasl_icp/build/test_results
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.14
-- BUILD_SHARED_LIBS is on
WARNING: Package "ompl" does not follow the version conventions. It should not contain leading zeros (unless the number is 0).
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 7 packages in topological order:
-- ~~  - ethzasl_icp_mapping (metapackage)
-- ~~  - ethzasl_gridmap_2d
-- ~~  - ethzasl_extrinsic_calibration
-- ~~  - libpointmatcher_ros
-- ~~  - ethzasl_icp_mapper
-- ~~  - ethzasl_icp_mapper_experiments
-- ~~  - ethzasl_point_cloud_vtk_tools
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'ethzasl_icp_mapping'
-- ==> add_subdirectory(ethzasl_icp_mapping/ethzasl_icp_mapping)
-- +++ processing catkin package: 'ethzasl_gridmap_2d'
-- ==> add_subdirectory(ethzasl_icp_mapping/ethzasl_gridmap_2d)
-- Using these message generators: gencpp;genlisp;genpy
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- checking for module 'eigen3'
--   found eigen3, version 3.2.0
-- Found Eigen: /usr/include/eigen3  
-- Eigen found (include: /usr/include/eigen3)
-- +++ processing catkin package: 'ethzasl_extrinsic_calibration'
-- ==> add_subdirectory(ethzasl_icp_mapping/ethzasl_extrinsic_calibration)
-- Using these message generators: gencpp;genlisp;genpy
[... 86 lines cut ...]
Scanning dependencies of target ethzasl_icp_mapper_generate_messages_lisp
[ 36%] Generating Lisp code from ethzasl_icp_mapper/GetMode.srv
[ 39%] Generating Lisp code from ethzasl_icp_mapper/SetMode.srv
[ 42%] Generating Lisp code from ethzasl_icp_mapper/LoadMap.srv
[ 45%] Generating Lisp code from ethzasl_icp_mapper/MatchClouds.srv
[ 48%] Generating Lisp code from ethzasl_icp_mapper/GetBoundedMap.srv
[ 51%] Generating Lisp code from ethzasl_icp_mapper/CorrectPose.srv
[ 51%] Built target ethzasl_icp_mapper_generate_messages_lisp
Scanning dependencies of target ethzasl_icp_mapper_generate_messages_py
[ 54%] Generating Python code from SRV ethzasl_icp_mapper/GetMode
[ 57%] Generating Python code from SRV ethzasl_icp_mapper/SetMode
[ 60%] Generating Python code from SRV ethzasl_icp_mapper/LoadMap
[ 63%] Generating Python code from SRV ethzasl_icp_mapper/MatchClouds
[ 63%] Built target tf_logger
[ 66%] [ 69%] Generating Python code from SRV ethzasl_icp_mapper/GetBoundedMap
Generating Python code from SRV ethzasl_icp_mapper/CorrectPose
[ 72%] Generating Python srv __init__.py for ethzasl_icp_mapper
Scanning dependencies of target ethzasl_icp_mapper_gencpp
[ 72%] Built target ethzasl_icp_mapper_gencpp
[ 72%] Built target ethzasl_icp_mapper_generate_messages_py
Scanning dependencies of target ethzasl_icp_mapper_generate_messages
[ 72%] Built target ethzasl_icp_mapper_generate_messages
make[1]: *** [ethzasl_icp_mapping/libpointmatcher_ros/CMakeFiles/pointmatcher_ros.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

nitin5, 2015-08-06:
Hi,
Here is output of catkin_make in verbose mode.

nitin@nitinHP:~/catkin_ethzasl_icp$ catkin_make VERBOSE=true
Base path: /home/nitin/catkin_ethzasl_icp
Source space: /home/nitin/catkin_ethzasl_icp/src
Build space: /home/nitin/catkin_ethzasl_icp/build
Devel space: /home/nitin/catkin_ethzasl_icp/devel
Install space: /home/nitin/catkin_ethzasl_icp/install
####
#### Running command: "make cmake_check_build_system" in "/home/nitin/catkin_ethzasl_icp/build"
####
####
#### Running command: "make VERBOSE=true -j4 -l4" in "/home/nitin/catkin_ethzasl_icp/build"
####
/usr/bin/cmake -H/home/nitin/catkin_ethzasl_icp/src -B/home/nitin/catkin_ethzasl_icp/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/nitin/catkin_ethzasl_icp/build/CMakeFiles /home/nitin/catkin_ethzasl_icp/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory `/home/nitin/catkin_ethzasl_icp/build'
make -f ethzasl_icp_mapping/ethzasl_gridmap_2d/CMakeFiles/ethzasl_gridmap_2d.dir/build.make ethzasl_icp_mapping/ethzasl_gridmap_2d/CMakeFiles/ethzasl_gridmap_2d.dir/depend
make -f ethzasl_icp_mapping/ethzasl_extrinsic_calibration/CMakeFiles/optimize.dir/build.make ethzasl_icp_mapping/ethzasl_extrinsic_calibration/CMakeFiles/optimize.dir/depend
make -f ethzasl_icp_mapping/ethzasl_extrinsic_calibration/CMakeFiles/tf_logger.dir/build.make ethzasl_icp_mapping/ethzasl_extrinsic_calibration/CMakeFiles/tf_logger.dir/depend
make[2]: Entering directory `/home/nitin/catkin_ethzasl_icp/build'
cd /home/nitin/catkin_ethzasl_icp/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/nitin/catkin_ethzasl_icp/src /home/nitin/catkin_ethzasl_icp/src/ethzasl_icp_mapping/ethzasl_gridmap_2d /home/nitin/catkin_ethzasl_icp/build /home/nitin/catkin_ethzasl_icp/build/ethzasl_icp_mapping/ethzasl_gridmap_2d /home/nitin/catkin_ethzasl_icp/build/ethzasl_icp_mapping/ethzasl_gridmap_2d/CMakeFiles/ethzasl_gridmap_2d.dir/DependInfo.cmake --color=
make[2]: Entering directory `/home/nitin/catkin_ethzasl_icp/build'
cd /home/nitin/catkin_ethzasl_icp/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/nitin/catkin_ethzasl_icp/src /home/nitin/catkin_ethzasl_icp/src/ethzasl_icp_mapping/ethzasl_extrinsic_calibration /home/nitin/catkin_ethzasl_icp/build /home/nitin/catkin_ethzasl_icp/build/ethzasl_icp_mapping/ethzasl_extrinsic_calibration /home/nitin/catkin_ethzasl_icp/build/ethzasl_icp_mapping/ethzasl_extrinsic_calibration/CMakeFiles/optimize.dir/DependInfo.cmake --color=
make -f ethzasl_icp_mapping/libpointmatcher_ros/CMakeFiles/pointmatcher_ros.dir/build.make ethzasl_icp_mapping/libpointmatcher_ros/CMakeFiles/pointmatcher_ros.dir/depend
make[2]: Entering directory `/home/nitin/catkin_ethzasl_icp/build'
cd /home/nitin/catkin_ethzasl_icp/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/nitin/catkin_ethzasl_icp/src /home/nitin/catkin_ethzasl_icp/src/ethzasl_icp_mapping/ethzasl_extrinsic_calibration /home/nitin/catkin_ethzasl_icp/build /home/nitin/catkin_ethzasl_icp/build/ethzasl_icp_mapping/ethzasl_extrinsic_calibration /home/nitin/catkin_ethzasl_icp/build/ethzasl_icp_mapping/ethzasl_extrinsic_calibration/CMakeFiles/tf_logger.dir/DependInfo.cmake --color=
make[2]: Entering directory `/home/nitin/catkin_ethzasl_icp/build'
cd /home/nitin/catkin_ethzasl_icp/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/nitin/catkin_ethzasl_icp/src /home/nitin/catkin_ethzasl_icp/src/ethzasl_icp_mapping/libpointmatcher_ros /home/nitin/catkin_ethzasl_icp/build /home/nitin/catkin_ethzasl_icp/build/ethzasl_icp_mapping/libpointmatcher_ros /home/nitin/catkin_ethzasl_icp/build/ethzasl_icp_mapping/libpointmatcher_ros/CMakeFiles/pointmatcher_ros.dir/DependInfo.cmake --color=
make[2]: Leaving directory `/home/nitin/catkin_ethzasl_icp/build'
make -f ethzasl_icp_mapping/ethzasl_extrinsic_calibration/CMakeFiles/tf_logger.dir/build.make ethzasl_icp_mapping/ethzasl_extrinsic_calibration/CMakeFiles/tf_logger.dir/build
make[2]: Leaving directory `/home/nitin/catkin_ethzasl_icp/build'
make[2]: Entering directory `/home/nitin/catkin_ethzasl_icp/build'
make[2]: Nothing to be done for `ethzasl_icp_mapping/ethzasl_extrinsic_calibration/CMakeFiles/tf_logger.dir/build'.
make -f ethzasl_icp_mapping/ethzasl_gridmap_2d/CMakeFiles/ethzasl_gridmap_2d.dir/build.make ethzasl_icp_mapping/ethzasl_gridmap_2d/CMakeFiles/ethzasl_gridmap_2d.dir/build
make[2]: Leaving directory `/home/nitin/catkin_ethzasl_icp/build'
make[2]: Leaving directory `/home/nitin/catkin_ethzasl_icp/build'
make -f ethzasl_icp_mapping/ethzasl_extrinsic_calibration/CMakeFiles/optimize.dir/build.make ethzasl_icp_mapping/ethzasl_extrinsic_calibration/CMakeFiles/optimize.dir/build
/usr/bin/cmake -E cmake_progress_report /home/nitin/catkin_ethzasl_icp/build/CMakeFiles  34
make[2]: Entering directory `/home/nitin/catkin_ethzasl_icp/build'
make[2]: Nothing to be done for `ethzasl_icp_mapping/ethzasl_extrinsic_calibration/CMakeFiles/optimize.dir/build'.
make[2]: Leaving directory `/home/nitin/catkin_ethzasl_icp/build'
/usr/bin/cmake -E cmake_progress_report /home/nitin/catkin_ethzasl_icp/build/CMakeFiles  30
make[2]: Entering directory `/home/nitin/catkin_ethzasl_icp/build'
make[2]: Nothing to be done for `ethzasl_icp_mapping/ethzasl_gridmap_2d/CMakeFiles/ethzasl_gridmap_2d.dir/build'.
make[2]: Leaving directory `/home/nitin/catkin_ethzasl_icp/build'
/usr/bin/cmake -E cmake_progress_report /home/nitin/catkin_ethzasl_icp/build/CMakeFiles  3
[... cut ...]
[  9%] [  9%] Built target std_msgs_generate_messages_cpp
Built target nav_msgs_generate_messages_cpp
make[2]: Leaving directory `/home/nitin/catkin_ethzasl_icp/build'
/usr/bin/cmake -E cmake_progress_report /home/nitin/catkin_ethzasl_icp/build/CMakeFiles 
[  9%] Built target _ethzasl_icp_mapper_generate_messages_check_deps_LoadMap
make[1]: Leaving directory `/home/nitin/catkin_ethzasl_icp/build'
make: *** [all] Error 2
Invoking "make VERBOSE=true -j4 -l4" failed

Have anyone tried compiling this code for indigo branch??

HannesSommer, 2015-08-07:
There were some fixes from master missing. I've just tried a reintegration. Could you please test the result? It is in branch ```reintegrate/master_into_indigo_devel```.

marinehorse, 2015-08-19:
Hi, 
I tried  git clone https://github.com/ethz-asl/ethzasl_icp_mapping.git -b reintegrate/master_into_indigo_devel

Still get the error:

Scanning dependencies of target _ethzasl_icp_mapper_generate_messages_check_deps_SetMode
[ 12%] Built target _ethzasl_icp_mapper_generate_messages_check_deps_SetMode
[ 15%] make[2]: *** No rule to make target `/tmp/buildd/ros-indigo-libpointmatcher-1.2.3-0trusty-20150517-2017/obj-x86_64-linux-gnu/libpointmatcher.so', needed by `/home/evan/workspace/ethzasl_icp_mapping_reintegraet_indigo/ethzasl_icp_mapping/devel/lib/libpointmatcher_ros.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....

HannesSommer, 2015-08-19:
Oha, something is still severely broken with the ROS Release of libpointmatcher. 
The quickest fix for you at the moment is probably to clone libpointmatcher (git@github.com:ethz-asl/libpointmatcher.git) into your catkin workspace's src folder and remove ros-indigo-libpointmatcher from your system. 
I'll try to fix the ROS release as soon as I find the time. (ros-indigo-libnabo, the major dependency from libpointmatcher should be fine that way).

marinehorse, 2015-08-26:
Hi @HannesSommer 
Thanks for your solution, now I can successfully build this package with libpointmatcher source.

nitin5, 2015-09-03:
I have also been able to compile the code using catkin_make_isolate .

closed 2015-09-03 by nitin5

HannesSommer, 2015-09-03:
Great and thanks for the feedback.