ethz-asl/ethzasl_icp_mapping #35: Make a release for Jade

pomerlef, 2015-08-18:


HannesSommer, 2015-10-15:
Okay we should urgently do that. We are currently loosing time because of that. A saw you, @pomerlef , merged the hydro_devel into the indigo_devel? does that mean that indigo_devel is now intended to run with hydro and indigo?

pomerlef, 2015-10-15:
That's my hope, but I'll need to install indigo in parallel to hydro on my machine. Otherwise, I'll to ask you everything if it's working and we will still be doing that by the end of your thesis ;)

The current status is that:
- the Indigo branch doesn't compile on Hydro (something related to `libpointmatcher_ros/include/pointmatcher_ros/ros_logger.h`)
- the Indigo branch doesn't compile on Indigo (most probably due to an API change I did and pushed half of it in the Indigo branch)

I'll try to have the Indigo branch compile. Once everything work on both sides, I'll check the difference between the branches.

pomerlef, 2015-10-15:
ok, the indigo branch is back to normal on Jenkins.

HannesSommer, 2015-10-15:
Okay great!

Here are some thoughts, just in case:
* Merging in master might help to avoid duplicate work, if you haven't done that already 
* Instead of merging master in indigo_devel it might be less effort to merge in my reintegrate/master_into_indigo_devel, where I did some of the necessary merging already recently.
* It might be easier and more representative to install Ubuntu 14.04 with indigo on some old or virtual computer instead of indigo on your 12.04. Maybe use nfs to share the source code.
* Alternatively you can use debootstrap and manual change root or use lxc to have a thin version of 14.04. living within your 12.04 - this would make it really easy to try out the same code on both systems as they can share files directly (a bit of work, but can be useful for many things).
* Just creating pull request to let Jenkins test for indigo compatibility might actually be the most cost effective way

baishibona, 2016-03-04:
Hey guys, just wondering if everything should work in Indigo now?
I downloaded the package and tried the steps on ROS wiki for compiling however got error with libnabo.
Thanks a lot for the help!

pomerlef, 2016-03-07:
It should work, but the instructions are outdated. You need to:

1. download the source for libnabo: https://github.com/ethz-asl/libnabo
1. `catkin_make_isolated ...`
1. download the source for libpointmatcher: https://github.com/ethz-asl/libpointmatcher
1. `catkin_make_isolated ...`
1. download this source
1. `catkin_make ...`

If that work for you, I'll update the documentation.

baishibona, 2016-03-08:
Hi François,

Thanks for the reply! I have tried the first two package which compile smoothly, however the third one doesn't go through. The error was "This workspace contains non-catkin packages in it", however if I try catkin_make_isolated, it won't compile the last package.
Should I keep everything in the same "catkin_ws"?
Thanks a lot!

pomerlef, 2016-03-09:
I think you need a special work space for the isolated packages and another
one for the standard packages.

François Pomerleau

Postdoctoral Researcher
UTIAS Robotics
University of Toronto, Ontario, Canada

*Google Scholar
<http://scholar.google.ca/citations?user=FJ9IqNoAAAAJ&hl=en>*
LinkedIn <http://www.linkedin.com/pub/fran%C3%A7ois-pomerleau/45/4a0/207>

baishibona, 2016-03-09:
I had the first two package in one catkin_ws1 and the last one in a separate catkin_ws2, however after source all the setup.sh I still ran into errors "ResourceNotFound: ethzasl_icp_mapper"
Any suggestions?
Thanks.

pomerlef, 2016-03-10:
So you have something like that:
```
[something]/catkin_ws1/src/libnabo
[something]/catkin_ws1/src/libpointmatcher

[something]/catkin_ws2/src/ethzasl_icp_mapper
```

and you did:
```
cd [something]/catkin_ws1/
catkin_make_isolated
```

and
```
cd [something]/catkin_ws2/
catkin_make
```

baishibona, 2016-03-10:
Yes, the only difference: "ethzasl_icp_mapping" was the package I
downloaded from github, if it is the same thing for "ethzasl_icp_mapper"

And then I tried to launch :
roslaunch ethzasl_icp_mapper tracker.launch

however getting the following error:
ERROR: cannot launch node of type [ethzasl_icp_mapper/mapper]: can't locate
node [mapper] in package [ethzasl_icp_mapper]

I am afraid I didn't get the ethzasl_icp_mapper compiling completely...

pomerlef, 2016-03-14:
humm I might have forget to tell you to use the hydro branch: https://github.com/ethz-asl/ethzasl_icp_mapping/tree/hydro_devel

The file `package.xml` is missing from the master branch as it is for Fuerte.

baishibona, 2016-03-16:
Getting a bit progress with the branch you provided, however getting error
with ethzasl_gridmap_2d which "Eigen" needs to be changed to Eigen3
However the following error I got no idea how to correct:

In file included from
/home/bona/ws2/src/ethzasl_icp_mapping-hydro_devel/ethzasl_icp_mapper/src/dynamic_mapper.cpp:19:0:
/home/bona/ws2/src/ethzasl_icp_mapping-hydro_devel/libpointmatcher_ros/include/pointmatcher_ros/ros_logger.h:
In member function ‘void
PointMatcherSupport::ROSLogger::writeRosLog(ros::console::Level, const
char*, int, const char*, const string&)’:
/home/bona/ws2/src/ethzasl_icp_mapping-hydro_devel/libpointmatcher_ros/include/pointmatcher_ros/ros_logger.h:55:27:
error: ‘loc’ was not declared in this scope
    ros::console::print(0, loc.logger_, loc.level_, file, line, func, "%s",
text.c_str());

This is actually the same one in "*vtk-tools" package.

Thanks in advance!

baishibona, 2016-04-02:
Hi Francois,

Sorry to bother, however any suggestion on the ros-indigo version?
Thanks!

./Bona

2016-03-16 1:16 GMT-04:00 shi bai <baishi.bona@gmail.com>:

pomerlef, 2016-04-27:
This error was reported by somebody else today. There seems to be an API difference between hydro and indigo. 

This is the problematic line:
https://github.com/ethz-asl/ethzasl_icp_mapping/blob/hydro_devel/libpointmatcher_ros/include/pointmatcher_ros/ros_logger.h#L55

and should be changed by:
https://github.com/ethz-asl/ethzasl_icp_mapping/blob/indigo_devel/libpointmatcher_ros/include/pointmatcher_ros/ros_logger.h#L54

@HannesSommer do you think it is possible to have those two API changes in a common branch?

We are moving to Jade on my side so I might invest some time into the branch feature/catkin_version to merge all of my latest work there and remove the other branches.

HannesSommer, 2016-04-27:
I don't know about a specific solution to this problem but it should be possible to find out at which ros version this changed and then use the `ROS_VERSION_MINIMUM` like in the example on http://answers.ros.org/question/40788/bullet_migration-electric-and-fuerte-compatibility/

baishibona, 2016-04-27:
@pomerlef Thanks, however I did find catkin_version works for ROS indigo a few days ago. So maybe suggest others to use catkin_version with the two dependencies you mentioned before should work.
Hopefully catkin_version will work for both Jade & Indigo.
Thanks again for the help and it is really nice package as we get pretty nice registration results over hundreds of meters mapping.

pomerlef, 2016-04-29:
It is actually harder than expected to find ROS versions in terms of number. The best I could do is check the [source of roscpp](https://github.com/ros/ros_comm/blob/kinetic-devel/clients/roscpp/package.xml), but Jade is not even listed in their branches.

- Hydro -> ROS_VERSION_MINIMUM(1, 10, 0)
- Indigo -> ROS_VERSION_MINIMUM(1, 11, 0)
- Jade -> ROS_VERSION_MINIMUM(1, 12, 0) __not clear__
- Kinetic -> ROS_VERSION_MINIMUM(1, 12, 0)

@HannesSommer I will slowly merge the modifications I did in the branch Hydro_dev in feature/catkin_version through this month. It's mainly a new way to handle time in libpointmatcher. Once this is done, I'll delete the other branches and point the master to catkin_version directly.

HannesSommer, 2016-04-30:
Oh, that is a bit sad - finding these numbers should not be hard!

But what you found is a good starting point for the following procedure, only good if there are only a few problematic spots in the code:
For each particular problem start with an initial guess for the critical transition version based on the list you've just produced and improve based on compiler feedback (ideally documenting best known upper and lower bounds as comments).
From there one can improve iteratively: As soon as a problem occurs with some untested version one bound and  the current best guess can be improved. I guess this converges very quickly and is less work than any code digging based alternative - sadly! 

Thanks for the update about your branch plans. Maybe you want to tag the old branches head's before you delete them. Otherwise you could actually loose some historic commits upstream - but maybe that is a good thing?

pomerlef, 2016-05-02:
@FabioDubois what is the motivation to link to an internal version of [Eigen](https://github.com/ethz-asl/eigen_catkin)? This make it tedious for external users to install the stack.

[This fail](https://github.com/ethz-asl/ethzasl_icp_mapping/blob/feature/catkin_version/libpointmatcher_ros/package.xml#L25) on my side:
```
ethzasl_icp_mapper: Cannot locate rosdep definition for [eigen_catkin]
libpointmatcher_ros: Cannot locate rosdep definition for [eigen_catkin]
ethzasl_gridmap_2d: Cannot locate rosdep definition for [eigen_catkin]

```

In general, what is the difference between the branches:
```
reintegrate/master_into_indigo_devel
feature/catkin_version
```

HannesSommer, 2016-05-10:
The `reintegrate/master_into_indigo_devel` I created back then to help in (#34) to use the master with indigo. It also helped in #37. I left it for the case that it would become useful for whoever would do the same work in a less temporary way. I don't believe it can help any more from here on. Or maybe it can?

I think the reference to `eigen_catkin` should not be in there. This is too specific. 
If I remember right we discussed how to depend on Eigen quite intensively in #38. 
@FabioDubois , can you find an alternative solution in there? If you don't have the time, I could also try to sort this out.

FabioDubois, 2016-05-11:
@pomerlef @HannesSommer The intention for opening the branch `feature/catkin_version` (out of `master`) was to use the ICP mapper functionality in a catkin environment, which is working for us.

In order to keep using our branch until a new catkin-supporting ethzasl_icp_mapping release (for newest ROS versions) is available, should we rename our branch to avoid confusion for externals?

mehditlili, 2016-05-11:
Having compiled everything successfully, and installing it as well using catkin_make install (and catkin_make_isolated --install for the two libs) running my launch file fails
```
core service [/rosout] found
/opt/workspaces/catkin_ws/install/lib/ethzasl_icp_mapper/mapper: error while loading shared libraries: libpointmatcher_ros.so: cannot open shared object file: No such file or directory
process[mapper-1]: starte
```
Any hints how to fix that? I am using the branch indigo-devel

gawela, 2016-05-11:
@mehditlili you should create a new issue for this.
Have you sources your workspace after compiling the library, i.e.,
`source /opt/workspaces/catkin_ws/devel/setup.sh` ?

mehditlili, 2016-05-12:
The thing is I am using the install folder /opt/workspaces/catkin_ws/install , I will try again with devel

pomerlef, 2016-05-12:
@FabioDubois thanks for the clarification, I was about to do changes in `feature/catkin_version`, but I will let it alone until we are done with the formal catkin port.

@HannesSommer I'll work on `reintegrate/master_into_indigo_devel` to make sure that everything is up-to-date for my current work and that it compile on Jade. This branch will be the new prospect for the single catkin branch.

yantramanava, 2016-05-20:
Hi,
I got everything working on the ROS jade using some simple tweaks on the feature/catkin_version. I have documented them on the ros wiki http://wiki.ros.org/ethzasl_icp_mapping?distro=hydro 
Please take a look and correct if necessary. Also thanks to everyone for keeping this updated.