ethz-asl/ethzasl_icp_mapping #37: Eigen includes

aleksandaratanasov, 2015-09-04:
Incorrect includes of Eigen inside **ethzasl_icp_mapping/ethzasl_extrinsic_calibration/src/optimize.cpp**.

aleksandaratanasov, 2015-09-04:
The includes should be something like:

#include <eigen3/Eigen/Eigen>
#include <eigen3/Eigen/LU>

depending on the version installed. In my case I have ROS Indigo and try to compile the catkanized icp_mapping packages. The fix I have posted in this comment is a possible solution to the issue.

HannesSommer, 2015-09-07:
No, this is wrong. The build system should add the ```eigen3``` as part of the include list. (see e.g. http://eigen.tuxfamily.org/dox/GettingStarted.html). Your PR #36 is missing this. It requires something like 
```
include_directories(${EIGEN3_INCLUDE_DIR}) 
```

The ```catkinization``` branch is outdated.
The necessary changes are already within the ```indigo_devel``` branch.  Please try the ```reintegrate/master_into_indigo_devel``` branch if it works for you.

aleksandaratanasov, 2015-09-07:
Ah, you are right. I actually added the include_directories by myself but forgot about that. :D Will try your suggestion. Thanks!

closed 2015-09-07 by aleksandaratanasov