ethz-asl/ethzasl_icp_mapping #39: complie the ethzasl_icp_mapper error.( indigo_devel)

csc301, 2015-09-15:
/home/ycc/g2o_catkin/src/ethzasl_icp_mapping/ethzasl_icp_mapper/src/interact_mapper.cpp:446:61:   required from here
/usr/include/eigen3/Eigen/src/Core/util/StaticAssert.h:32:40: error: static assertion failed: THE_BRACKET_OPERATOR_IS_ONLY_FOR_VECTORS__USE_THE_PARENTHESIS_OPERATOR_INSTEAD
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
                                        ^
/usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:153:7: note: in expansion of macro ‘EIGEN_STATIC_ASSERT’
       EIGEN_STATIC_ASSERT(Derived::IsVectorAtCompileTime,

robnsngh, 2015-09-27:
The above error is raised when EIGEN2_SUPPORT is not declared in DenseCoeffsBase.h and consecutively the assert failed. I found a temporary way around it by commenting out the assert check.

pomerlef, 2015-10-08:
This should be fixed with the latest version.

closed 2015-10-08 by pomerlef