ethz-asl/ethzasl_icp_mapping #43: Any SLAM solution available here?

baishibona, 2016-05-02:
Hi,

I was wondering is there any consideration of a SLAM solution build on top of 3D scan matching?
Right now I am trying to put it together with iSAM which enable us to update the map by using key frames. I wasn't quite sure if you guys have developed that function or it is under developing?

Thanks.

pomerlef, 2016-05-03:
Right now, there is none. We manage to go pretty far without global relaxation, so it wasn't a priority.

Recent results:
https://www.youtube.com/watch?v=jlAArM_6N-I

baishibona, 2016-05-03:
That's a cool video, however we were using the puck, which is not as powerful as 64E, and we did have errors while the robot was in a forest like environment.
Could you point me to the function in ethzasl_icp_mapper that I can call for a scan matching between two point clouds (ideally ROS point cloud msg or just point cloud object)?

Thanks!

pomerlef, 2016-05-11:
The conversion of pointcloud2 to our format (essentially a big Eigen matrix) is done [here](https://github.com/ethz-asl/ethzasl_icp_mapping/blob/master/ethzasl_icp_mapper/src/dynamic_mapper.cpp#L425). It looks more complicated than how it could be used:
```
DP points = PointMatcher_ros::rosMsgToPointMatcherCloud<float>(cloudMsgIn);
```

The ICP part happen [here](https://github.com/ethz-asl/ethzasl_icp_mapping/blob/master/ethzasl_icp_mapper/src/dynamic_mapper.cpp#L566). The object `icp` used here hold internal a map and is from the class `ICPSequence` defined in [libpointmatcher](https://github.com/ethz-asl/libpointmatcher).

This repository is mainly supported for our research partners and our changing needs. We have less time for public support right now.

baishibona, 2016-05-12:
Thank you very much! I think I should be able to sort it out.

closed 2016-05-12 by pomerlef