ethz-asl/ethzasl_icp_mapping #44: Parameter prob for module MaxPointCountDataPointsFilter was set but is not used

mehditlili, 2016-05-12:
When starting mapper from the indigo-devel branch, using the openni launcher.
```
terminate called after throwing an instance of 'PointMatcherSupport::Parametrizable::InvalidParameter'
  what():  Parameter prob for module MaxPointCountDataPointsFilter was set but is not used
[mapper-1] process has died [pid 445, exit code -6, cmd /opt/workspaces/catkin_eth/devel/lib/ethzasl_icp_mapper/mapper cloud_in:=/pico_flexx/points __name:=mapper __log:=/home/mehdi/.ros/log/ba3bf752-183d-11e6-acb5-28d24422fddc/mapper-1.log].
log file: /home/mehdi/.ros/log/ba3bf752-183d-11e6-acb5-28d24422fddc/mapper-1*.log

```

pomerlef, 2016-05-12:
Find the yaml file using `MaxPointCountDataPointsFilter`

replace its parameter `prob` by `maxCount`. It is currently a mess in all the branches. We are working on a collapsing the branches to simplify the maintenance, but it will take a couple of months.

Which launch file were you using?

The-gimmickk, 2016-09-09:
I am getting the same Problem with every openni launch file can you please tell me where can I find MaxPointCountDataPointsFilter.yaml ?  I found defaultMaxPointCountDataPointsFilter.yaml in libpointmatcher source code but I dont think that is going to be useful, and even in that yaml file there's no prob parameter. 
Please help!
I am using Ubuntu 15.10 with kinetic distro of ROS.
This error pops out when i run tracker.launch.

pomerlef, 2016-09-11:
This error was fixed on the branch `indigo_devel`. What branch are you using? 

In general, we are slowly stopping support for this code and encourage people to build there own with `libpointmatcher` directly.

The-gimmickk, 2016-09-11:
I am using indigo_devel only, but I think the source code should work on ROS kinetic as well. 

Can you tell me where can i found MaxPointCountDataPointsFilter.yaml?