ethz-asl/ethzasl_icp_mapping #52: no rule to make target

MaMadDl, 2016-12-13:
Hi im getting following error when i try to compile under ros-indigo and catkin:
No rule to make target `/tmp/binarydeb/ros-indigo-libpointmatcher-1.2.3/obj-x86_64-linux-gnu/libpointmatcher.so', needed by `/home/mamaddl/catkin_ws/devel/lib/libpointmatcher_ros.so

plz help me thanks in advance.

pomerlef, 2016-12-13:
I'm not sure from where you got `libpointmatcher`. You should remove your current version and install it from source.

MaMadDl, 2016-12-13:
i got it with apt-get with name ros-indigo-libpointmatcher
when i remove it i get the dependency missing error:

Could not find a package configuration file provided by "libpointmatcher"
  with any of the following names:

    libpointmatcherConfig.cmake
    libpointmatcher-config.cmake

  Add the installation prefix of "libpointmatcher" to CMAKE_PREFIX_PATH or
  set "libpointmatcher_DIR" to a directory containing one of the above files.
  If "libpointmatcher" provides a separate development package or SDK, be
  sure it has been installed.
Call Stack (most recent call first):
  ethzasl_icp_mapping/libpointmatcher_ros/CMakeLists.txt:5 (catkin_simple)

HannesSommer, 2016-12-13:
This package is probably still not installing properly. For the moment, @MaMadDl , please remove it (`apt-get remove ros-indigo-libpointmatcher`) and clone libpointmatcher (`https://github.com/ethz-asl/libpointmatcher.git`) into your catkin workspace or do a system installation from the same source.

MaMadDl, 2016-12-13:
thanks problem solved with installing from source

closed 2016-12-13 by pomerlef