ethz-asl/ethzasl_icp_mapping #53: ICP Sequence Mapping

ArthurVanpoucke, 2016-12-14:
Hi, thank you for all your work it has been very usefull.

I'm trying to build a point cloud map from scans and I'm experiencing some troubles with ICP algorithm. 

 I've tried to understand your code examples like dynamicmapper.cpp but there is some operations that i don't understand. 

So there is my method and i don't know if it's correct to think that way.

Say we loop for 10 pointclouds, i compute a globalTransformation (that is the expression of the initial frame from the actual pointcloud frame) by multiplying matrix given by ICPSequence::icp(newCloud,globalTransormation). 
In my code i have somethign like : 

ICPSequence icp
PM::TransformationParameters globalTransformation;
PM::TransformationParameters T;

Then when I perfom icp

T = icp(newCloud, globalTransformation);
globalTransformation = globalTransformation * T;

Then i do something like this : 

 PM::Transformation* rigidTrans;
 rigidTrans = PM::get().REG(Transformation).create("RigidTransformation");
 rigidTrans->compute(newCloud,globalTransformation)

mapPointcloud.concatenate(newCloud);

And i concatenate to a PointCloud all the transformed pointcloud.

Do this method seems right to you ?

Thank you in advance for your answer

Arthur

pomerlef, 2016-12-14:
Hi Artur,

We try to reduce the support to a minimum when it comes to specific application. As we have a limited time for support, we prefer to put it in solving bugs or compilation issues.

That being said, you might want to use `PM::ICP icp` as it is simpler to use. With `PM::ICPSequence icp', you need to separatly provide the reference point cloud with a call to `icp.setMap(mapPointCloud)`. This save computational time if you don't want to merge all the new point clouds.

Here is a pseudo-code that might help you:
```
PM::ICP icp;
icp.loadFromYaml(<your file with icp configuration>);

PM::Transformation* rigidTrans;
rigidTrans = PM::get().REG(Transformation).create("RigidTransformation");

PM::TP globalTransformation = PM::TP::identity(4,4);

while(new point coming)
{
  if(first point cloud)
  {
    mapPointcloud = newPointCloud;
    continue;
  }
  
  // if you have a better first guess that the last transformation, you can use it here
  PM::TP T = icp(newPointCloud, mapPointcloud, globalTransformation);
  globalTransformation = globalTransformation * T;

  rigidTrans->compute(newCloud,globalTransformation);
  mapPointcloud.concatenate(newCloud);
}

// so you can inspect your map in paraview for example
mapPointcloud.save("myGlobalMap.vtk");
```

ArthurVanpoucke, 2016-12-14:
Thank you for your fast answer

I'll test that as soon as i can !

ArthurVanpoucke, 2016-12-15:
When I try it that way i have an error with the line :
` rigidTrans->compute(newCloud,globalTransformation);`

terminate called after throwing an instance of 'PointMatcherSupport::TransformationError'
what(): RigidTransformation: Error, rotation matrix is not orthogonal

I tried to correct this with 
`rigidTransformation->checkParameters(globalTransformation) ` 
and 
`rigidTransformation->correctParameters(globalTransformation) ` 

but i getthe same error...
Do you have any idea of why i'm getting this ? I'm stuck with this problem of not orthogonal matrix and i dont understand where it comes from.

Thank you very much for your help and again for your time i know its not bug or compliation issues, do you know where can i ask my question ?

pomerlef, 2016-12-15:
What is your matrix?

```
std::cout << globalTransformation << std::endl;
```

ArthurVanpoucke, 2016-12-15:
My matrix in the n-1 process is :

```
global transformation :
    0.999835    0.0125031   -0.0277789     0.477889
  -0.0125225      1.00013 -0.000271116   -0.0520651
   0.0276331  0.000629483      0.99977   -0.0343972
           0            0            0            1
```

My matrix from ICP with `T = icp(newCloud, mapPointcloud, globalTransformation) : `

T :
```
     1.00029   0.00272306  -0.00131827     0.063286
 -0.00272871       1.0002  0.000168823  -0.00451348
  0.00117483 -0.000154402      1.00016 -6.85304e-05
           0            0            0            1
```

Then i update globalTransformation :

 `globalTransformation = globalTransformation *  T;`

And when i get to the line : 

`  rigidTrans->compute(newCloud,globalTransformation);
`

I get the error with the globalTransformation matrix which is : 
globalTransformation :
```
     1.00006    0.0152325   -0.0290991      0.54111
  -0.0152555      1.00029 -8.58063e-05   -0.0573716
    0.028814   0.00055049     0.999889   -0.0327198
           0            0            0            1
```

Maybe something to do with the frame of the pointclouds ? I have some doubts about this, my pointclouds should be all in the same local frame right ?

pomerlef, 2016-12-15:
You are on the in the noise range. The two first matrices are under the threshold, but not when they are combined.

This check was added to avoid drift when multiplying matrices in a loop, which eventually scale your point clouds on the long run. In other words, it's good that you catched that now, otherwise it would have been a mess to debug later.

To force the orthogonality, you need to use
```
globalTransformation = rigidTransformation->correctParameters(globalTransformation);
```

I think you were only missing the reassignment.

ArthurVanpoucke, 2016-12-15:
I added the line and i still got the same error like the correction didn't work : 

N-1 Process :

```
globalTransformation
  0.998737  0.0704271  0.0368053     1.4755
-0.0326875   0.996784 -0.0409052 -0.0960057
-0.0381537  0.0382395   0.998485 -0.0458018
         0          0          0          1
```

Before correction : 

```
globalTransformation :
  0.998991    0.10357  0.0356541    1.59347
-0.0295089   0.994979 -0.0449927  -0.102487
-0.0371571  0.0411058   0.998351 -0.0452284
         0          0          0          1

T from icp 

    0.999995    0.0330705 -0.000861123       0.1181
  0.00322462     0.999444   -0.0041293  -0.00241642
  0.00087449   0.00415558     0.999991   0.00517955
           0            0            0            1
```

globalTransformation = rigidTransformation->correctParameters(globalTransformation);

After correction : 

```
globalTransformation :

  0.993996  0.0311284  0.0356541    1.59347
 -0.101811   0.998551 -0.0449927  -0.102487
-0.0400869    0.04389   0.998351 -0.0452284
         0          0          0          1
```

Return the same error

pomerlef, 2016-12-15:
Are you doing the correction just before applying the transformation, after concatenating the matrices?

I would expect your corrected transformation to have larger translation components.

I updated the example code to be sure we are doing the same thing.

ArthurVanpoucke, 2016-12-15:
Yes i do the same thing, applying correction before the concatenation

pomerlef, 2016-12-15:
ok, I recomputed manually what the correction should give and there might be a bug in the function `correctParameters()`.

To give you a bit of context, we check if the determinant of the rotational part is very close to 1. In your case:
 - the original globalTransformation is 1.00131
 - the output of ICP is 1.00065
 - the combination globalTransformation*T is 1.00197

Anything above 1.001 will trigger the error.

Anyway, a call to `correctParameters()` should reduce this value, which doesn't seems to happen. I'll fix the error I saw in libpointmatcher and get back to you.

This did turned out to be a bug after all ;)

pomerlef, 2016-12-15:
Alright, you can go to your libpointmatcher repo and to `git pull` and `sudo make install`.

There was a unit test for the function `correctParameters()`, but with a very basic non-orthogonal matrix. I put one of your matrices there, which was more tricky to correct.

ArthurVanpoucke, 2016-12-15:
Thank you for your rapidity ! I will take a look at it and tell you what i get :)

pomerlef, 2016-12-15:
I also updated our example code for a sequence of scans: https://github.com/ethz-asl/libpointmatcher/blob/master/examples/align_sequence.cpp

I think you do not need to compose the transformation together.

ArthurVanpoucke, 2016-12-20:
Thank you very much, it's working now !!!

closed 2016-12-20 by ArthurVanpoucke