ethz-asl/ethzasl_icp_mapping #54: Compiling for ROS Kinetic

mgb45, 2017-02-21:
I'm not sure what branches are meant to do what right now, whether `feature/catkin_version` or `indigo_devel` is the current recommended version, so haven't sent a pull request, but the branch [here](https://github.com/mgb45/ethzasl_icp_mapping/tree/kinetic) compiles for ROS K.
 
I based this on `indigo_devel`, and simply added `include_directories(include ${Eigen_INCLUDE_DIRS} ${catkin_INCLUDE_DIRS})` to the `ethzasl_gridmap_2d/CMakeLists.txt`

Compiling requires:
```
mkdir -p catkin_ws/ws1/src
cd catkin/ws1/src
catkin_init_workspace
git clone https://github.com/ethz-asl/libnabo.git
git clone https://github.com/ethz-asl/libpointmatcher.git
cd ..
catkin_make_isolated
source devel_isolated/setup.bash

cd ..
mkdir -p ws2/src
cd ws2/src
https://github.com/mgb45/ethzasl_icp_mapping.git
git checkout kinetic
cd ..
catkin_make
```

pomerlef, 2017-02-21:
Sorry, it's a mess right now in terms of branches. The most up-to-date one is `reintegrate/master_into_indigo_devel`. Could you check if your patch work on that branch too?

jcarius, 2017-05-04:
I tried today and reached the same conclusion. Compiles fine with the additional `${catkin_INCLUDE_DIRS}` .
Please merge!

adaltun, 2019-06-20:
I am trying to launch any file however each of them gives error of :

ERROR: cannot launch node of type [ethzasl_icp_mapper/mapper]: can't locate node [mapper] in package [ethzasl_icp_mapper]

I have, double, triple checked my workspace however cant find a solution. Did anyone had the same problem and could solve it in a specific way, I will appreciate if you share it.

closed 2019-07-09 by mgb45