ethz-asl/ethzasl_icp_mapping #56: Any update on using this with Indigo?

bobjohnson2040, 2017-04-23:
Hello,

Do you have any update on getting this to be compatible with Indigo? I tried catkin_make_isolated on the master branch but the 'mapper' node was missing.

pomerlef, 2017-04-24:
You need to compile `libpointmatcher` and `libnabo` with `catkin_make_isolated` not this package. 

Once both of those libraries are compiled, you can use the branch `reintegrate/master_into_indigo_devel` and it should be fine.

bobjohnson2040, 2017-04-27:
Thank you very much for the quick response! I really appreciate all of the hard work that you have put into building this library and for taking the time to answer my questions! This package looks awesome, you did an amazing job building it!!

What is the difference between the branch `reintegrate/master_into_indigo_devel` and `indigo_devel` ? 

After spending a bunch of time messing with all of the different branches the day after I opened this issue, I was able to get the `indigo_devel` branch to work but not `reintegrate/master_into_indigo_devel`. I am using Ubuntu 14.04 and ROS Indigo with a Velodyne VLP-16 LIDAR. 

I will try again tomorrow to see if I can get the `reintegrate/master_into_indigo_devel` branch to work on my system. If I'm able to get it working, then would you accept a pull request to update the README with instructions?

pomerlef, 2017-04-27:
Essentially, I don't recall what is in `indigo_devel`. The branches are a mess since a couple of years. We are slowly depreciating the support for the node.

When I fix or add something, it's in `reintegrate/master_into_indigo_devel`. 

Any documentation help is appreciated!

bobjohnson2040, 2017-05-22:
Do you guys currently use a different open source SLAM software for ROS instead of this one? I've been trying to find a good one to use with a Velodyne LIDAR

anand-bala, 2017-07-25:
@bobjohnson2040 Do you have any updates on a good SLAM package to use with a Velodyne LIDAR?