ethz-asl/ethzasl_icp_mapping #60: ERROR: cannot launch node of type [point_cloud_converter/point_cloud_converter]: point_cloud_converter

mjjdick, 2017-09-20:
Hi:
    when I try to  roslaunch ethzasl_icp_mapper ICRA14_dynamic_mapper.launch, I just get this error!
    so I download a package named point_cloud_converter, it still can run right. besides, in the package ,I did not found the node named point_cloud_converter_obs. are there anyone can help me? thx very much!

pomerlef, 2017-09-20:
I'm not sure in which branch you are but make sure to use `reintegrate/master_into_indigo_devel`

mjjdick, 2017-09-21:
thx for your response! I am in reintegrate/master_into_indigo_devel branch, and the launch file path is ethzasl_icp_mapping/ethzasl_icp_mapper/launch/artor/ICRA14/ICRA14_dynamic_mapper.launch.