ethz-asl/ethzasl_icp_mapping #62: support for Velodyne HDL-32e?

Chambana, 2017-10-19:
The launch files have support for the 64-laser Velodyne and the Velodyne puck, but no explicit option to use the HDL-32e.

pomerlef, 2017-10-19:
Look for the launch files in the folder Artor. That robot was equipped with a HDL-32e

Chambana, 2017-10-19:
Terrific.  Do you remember if the Artor setup required a separate IMU for SLAM?  I only have the Velodyne data....

liuqi8827, 2019-02-13:
@pomerlef Hi,
Dose the package support for VLP-16 (Velodyne 16)?

best

pomerlef, 2019-02-14:
The node support anything producing a pointCloud2 message. How you tune it to work, that's up to you.

You can have a look at `ethzasl_icp_mapper/launch/velodyne_puck/` as a starting point.

liuqi8827, 2019-02-15:
@pomerlef 
I've got it.

Tanks for your kind and prompt reply.

closed 2019-02-15 by pomerlef