ethz-asl/ethzasl_icp_mapping #67: Real time processing

swarmt, 2018-01-04:
Thanks for the excellent library. 
I am getting a lot of warnings about real time perfomance using the vlp16 launch files and filters. Could you please point me in the right direction for which filters to tune with the best performance increase and least accuracy lost?

pomerlef, 2018-01-04:
That depends a lot on your application and your processing power. For slow motions, skipping few scans is not a problem. 

Typically, reducing the number of points helps. Random sampling is fast to compute, but it's blind to the structure of your scan. Density filtering is slower, but will yield more stable results with fewer points.

swarmt, 2018-01-04:
Great thanks. I will tinker with those!

closed 2018-01-04 by swarmt