ethz-asl/ethzasl_icp_mapping #70: No module error python node

BalasubramanyamEvani, 2018-06-20:
When I try to run the node in the ethzasl_icp_mapper/nodes/ , it shows me an error stating no module found 
at import modular_cloud_matcher.srv and at import point_cloud.

pomerlef, 2018-06-20:
This is an old node that was not supported since 6 years. Use `dynamic_mapper.cpp` instead.

BalasubramanyamEvani, 2018-06-20:
Thank you for replying. 

I am not well versed with roscpp, I created a point cloud of a map and wanted to use the subsequent scans that are coming from the 2D LIDAR and register it to the map, I am publishing both the map and scans that are coming, in the form point cloud messages. can it be done? I am not sure what changes may need to be done if you would kindly help me with this, I mostly use python, I have gone over the ros tutorials for roscpp but haven't programmed much in c++.   

Thanks

BalasubramanyamEvani, 2018-06-21:
Hello,
Is there any method by which I can register a pair of point clouds. The point clouds are coming from two different topics.